OpenEVSE/OpenEVSE_PLUS #2: Bill of materials not ok?!

micdre, 2017-04-11:
Hello,

I found out that the BOM list, genretated in Eagle, is not ok in version 4 and 3! (maybe other versions too)
Is there somewhere a good list.

Regards, Michel

chris1howell, 2017-04-11:
I will take a look at the Eagle generated BOM. Any information on what parts are not okay?

micdre, 2017-04-11:
Hello
Thanks!

IC2 is not a mega 8 but a 328P
IC3 is not a LMV358 but a LM358
IC4 is not a LMV358 bu a LF353
Maybe there's more...

Costs me a couple of parts....

Regards, Michel

Verstuurd vanaf mijn iPhone

chris1howell, 2017-04-11:
I updated the following part values so generic footprints report the correct part when exporting Eagle generated BOM.

IC2 value 328P 
IC3 value LM358 
IC4 value LF353
TVS value P6SMB16CA
D1 value AZ23C5V6-7-F

closed 2017-04-11 by chris1howell

micdre, 2017-04-11:
In wich version of the board?

Verstuurd vanaf mijn iPhone

chris1howell, 2017-04-11:
V4.

micdre, 2017-04-11:
Thanks!

Verstuurd vanaf mijn iPhone

V4.