OpenEVSE/OpenEVSE_PLUS #5: Math / Theory of the GFCI circuit

RyanCopley, 2019-05-25:
I did find the theory of operations, and I reviewed the burden resistor calculator, but I am at odds with trying to understand how a burden resistor of 412 ohms was selected. 

For the CT being used (http://www.crmagnetics.com/Assets/ProductPDFs/CR8400%20Series.pdf) it has the specs
Max RMS Current / Ir = 20 amps
Current Sensor turn ratio / Te = 1011:1
The arduino System Voltage is 5v

I plug these values in to https://tyler.anairo.com/projects/open-energy-monitor-calculator and I get a burden resistor of 89 ohms. Are you scaling down the max RMS current to ~4.3 amps for more sensitive detections?

Could you also go into explaining what the first OP Amp is for? I understand why the second one exists, but the first is elusive to me.

I would really appreciate the help understanding the theory here :-)

chris1howell, 2019-05-26:
The coil is scaled way down, there is no point reading anything over 20ma.
Sensitivity to low currents is more important.

The first stage of the Opamp is a amplifier or multiplyer. The input which
is a very tiny voltage is amplified before hitting the second stage which
is a voltage comparator. If the amplified voltage from the coil is above
the threshold the output of the opamp goes high. The Inturupt on the
microprocessor is triggered on the rising voltage of the opamp output.

RyanCopley, 2019-05-26:
Thank for you for the explanation, that is really helpful!

closed 2019-05-26 by RyanCopley