OpenEVSE/OpenEVSE_PLUS #8: Non OpenESVE LCD

samigroup, 2019-08-08:
Hello, 
I am experiencing around with the OpenEVSE Plus Controller, I have added my available LCD 16X2, with pack module for I2C is there any way to update the I2C address without flashing the OpenEVSE controller code? 
-- I know the LCD through the OpenEVSE is jumped A0-A2, so I soldered the A0-A2 jumpers to get the same address,  but its not just that, its more, where its two different communication schemes. But its protected for throught the header files and headers, I was wondering if I can change the selection of LCD definition/I2C header through the RPI interface (wifi interface)  
 
- I am not using the RTC, nor the Temperature, ( disabled them through wifi interface)

samigroup, 2019-08-26:
If anyone was successful with this, it would be great if you can give me some pointers. Thanks!

chris1howell, 2019-08-26:
To use a different LCD/Controller combo you will need to modify the source
code to support your particular device and recompile.