OpenEVSE/OpenEVSE_PLUS #10: Suggestion for PP

giorgio80, 2020-01-20:
Dear,

I'm trying to maker hobby a board but it's  It's not clear pp specification. I better explain.  
In particular,  it's clear for me how pp work,  simply reading resistance state to understand supported cable power and car presence,  but is not clear if it has to be 5v for specification. 

I better explain. 
I see it is connected to 5v then to add for reading inout.  My board works on 3.3v   so my question is :  would 3.3v setup would work or pp has to be 5v for j1772 specific ? 

thanks.

sauttefk, 2020-01-20:
You can also use just 3.3V for your circuit. You just have to measure the resistance.
5V was the standard voltage of digital circuits back in the time, J1772 was standardized.

But 5V will have a higher noise immunity than just 3.3V, but i don't think that matters, as the resistance to measure is low enough (the current high enough).

Make sure to protect your PP input circuitry against static discharge and  leakage current due to wet/dirty ev-connector/short circuit to CP.

giorgio80, 2020-01-21:
Thanks for your answer, really more clear now . About checking the value can I also ask you when in the charging process it has to be checked ?
Thanks again :)

sauttefk, 2020-01-22:
You are developing a device that has a lot of inherent life threatening electrical safety risks!
Of course you have to monitor all values continuosly and shut off the power immediately when any parameter exceeds acceptable limits.
Maybe you should invest in getting the IEC61851 standard from ISO.

giorgio80, 2020-01-23:
hi @sauttefk 
sorry I was not clear.  it's clear this has to be monitored continuously especially for overcurrent, real question was when PP has to be checked initially ,  I suppose immediately after pilot start ( and adapting load consequently) .  is this a right approach? I read around some other say that no Pwm pilot has to be started before pp is checked other instead say to start Pwm and continuously check for pp and to stop (without open contactor) charge process if no certified cable is detected.

Niko-La, 2020-01-30:
Make sure to protect your PP input circuitry against static discharge and leakage current due to wet/dirty ev-connector/short circuit to CP.

@sauttefk can you please elaborate on this.  Does openevse already do this in the schematic.

Also where can I follow the work being done for OpenEvse v6