OpenEVSE/OpenEVSE_PLUS #11: Incorrect GFCI self-test current?

davidrea, 2020-01-22:
The V5 schematic has a 330Ω resistor and 5V drive signal, resulting in a 0-15.15mA square wave through the self-test winding. The [CT product page](https://store.openevse.com/collections/all-products/products/gfci-current-transformer) indicates a trip current of 15-20mA, and a 5-turn self-test coil. 

Should the resistor on the board actually be 1.6KΩ so that you only get 3mA through the self-test primary winding? In this case the voltage seen from the GFCI CT would be 5x that, since the nominal "pass the load wires through" case is a 1-turn primary. This would run the self-test at a current transfer ratio much closer to the design-intent in-use threshold.

(It's also understandable that there could be some CTR loss due to the crest factor of the square wave drive signal; the closest-to-ideal CTR would be with a sinusoidal input, so this might need to be derated a bit.)

chris1howell, 2020-01-22:
The UL current maximum for the test is 30ma. The test signal is not solid DC but PWM, the duty cycle is set to stay under the maximum.

UL has tested this setup with Wattzilla in the OpenEVSE v4 and are satisfied with the theoritialcal and tested levels.

Chris

davidrea, 2020-01-23:
Excellent, thanks for the clarification!

closed 2020-01-23 by davidrea