OpenEVSE/OpenEVSE_PLUS #12: GFCI tripping point.

faustomilletari, 2020-09-30:
Dear community,

According to this thread https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwjLspSUrJDsAhUyyoUKHVPdCcAQFjABegQIBxAB&url=https%3A%2F%2Fcommunity.openenergymonitor.org%2Ft%2Fopenevse-and-wiring-regs-changes-722-531-2-101%2F10959&usg=AOvVaw3PmfsgyS8OJmghcbeV4KOB 

there are changes in the regulation that require the GFCI to be more sensitive to current leakages. It seems that the current OpenEVSE hardware is built to have the GFCI trip at 15mA, whereas regulation requires 6mA. 

I understand that changing the tripping point requires changes in the resitors used as burden resistor etc. I would like to ask you if 
1) you think the current circuit can be adapted to the new requirements by swapping resistors
2) the new 6mA limit is high enough not to cause any issues with false positive tripping signals

I would like to state that i'm not an expert, and therefore i am still learning a lot. if i have misstated something in this issue please let me know and feel free to close it. 

If also you could help me make calculations for new resistor values for the GFCI, in the sense of pointing me to material that can help me understand how that circuit works, i would be very grateful!

sauttefk, 2020-09-30:
**Caution:**
The current transformer used with OpenEVSE does not protect from **DC** fault currents.
Due to the fact that it is a coil around the conductors it can only pick up **AC** currents.

AC tripping current has to be at most 30mA
DC tripping current has to be at most 6mA

To achieve this you either need a Type B GFCI or a Type A GFCI plus something like the [WA RCM14-03 AC/DC Residual Current Monitor](https://deltrixchargers.com/wp-content/uploads/2019/10/WA-DS-015-RCM14-03-Rev-B.pdf).

A [hall effect sensor](https://en.wikipedia.org/wiki/Hall-effect_sensor) - in contrast to a current transformer coil - can detect DC current.

faustomilletari, 2020-09-30:
I am confused as of why this is needed at all... For a "normal" single or three-phase charging station where would we see this DC current coming from? The only place with DC current is the board itself and the pilot. 

is this regulation meant to be applied by DC charging stations?

sauttefk, 2020-09-30:
A DC ground fault can occur, when a ground fault occurs behind the rectifier i.e. a battery pole has connection to earth.

faustomilletari, 2020-09-30:
Yes. now i understand. I am reading about it in this interesting article which says exactly what you stated in your last answer. http://www.mastrogippo.it/2019/09/rcm-direct-current-sensing-how-its-done/ 

thank you for helping me. It seems like this system is not required for EVSE meant to be home chargers for private use. So i think it might not be needed in my case... 

thanks

SDCSolutions, 2020-09-30:
Looks like it is for the IEC62196 standard which governs CCS or DC charging. Also hence the loss of ground detect in the spec. http://www.doepke.co.uk/download/Techpub-06.pdf

sauttefk, 2020-09-30:
**Attention:**
A DC ground fault can also occur with AC charging. A connection of the HV-battery to the chassis can easily happen e.g. a marten chewing on cables or rubbed through cables.

SDCSolutions, 2020-09-30:
So would that type of RCD need to be mounted on the ground line?

sauttefk, 2020-09-30:
An affordable Type B RCD is this [GEYA GYL9 63A 30mA](https://de.aliexpress.com/item/33020060308.html)
An RCD is your life insurance when something bad happens to your EVSE / cable / connector / car in this wet, hot, cold, dirty environment you are going to deliver deadly voltages.

sauttefk, 2020-09-30:
When you do a 240V 2phase charging in North America, a 4 pole RCCB has to be used, leaving on pole unconnected (N/L1/L2/nc).

SDCSolutions, 2020-09-30:
Frank,
I guess my question is how is an RCD supposed to protect against a vehicle fault. Most of these control systems are designed to protect against unwanted electricity leaving the EVSE via the plug.

If I am understanding you correctly, this legislation is designed to protect against energy leaving the car?

Or is this in the event of a poorly set up solar panel installation?

sauttefk, 2020-09-30:
A RCD monitors if any current sent out on one conductor does not return on any other conductor, but goes another way to earth. In the worst scenario through your heart.
So any insulation fault behind causing a current to earth, trips the RCD and therefore protects life and property.

A RCD can also be used to automatically isolate a faulty solar panel installation.

SDCSolutions, 2020-09-30:
Hi Frank,
I’m just trying to identify the source of the hazard here, the device that is being protected and the rationale of using it in a level 2 product.

If the source of the hazard is the vehicle, the RCD in the EVSE  is designed to protect the house?

I can understand the RCD being necessary for a solar panel, but then I question the rationale behind making the legislation around the EVSE as opposed to the solar panel.

If the hazard is the utility sending DC power,  along the same lines as the AC power, then I guess I would consider that a potentially valid risk. Not sure how it would happen as substations are usually all transformers where I live and it’s a little difficult to pass DC through them, but if I want to be protected from bad actors or negligent installations, why not make every plug an RCD circuit and leave the expensive hardware out of the EVSE.

Fundamentally, my question is why is this necessary to be put on a product people are already complaining is too expensive?

sauttefk, 2020-10-01:
The International Electrotechnical Commission Standard [IEC 60364-7-722](https://webstore.iec.ch/publication/29958) demands either a 30mA Type B RCCB or a special Type EV (30mA AC/6mA DC) RCCB.

I don't know the legal requirements all over the world. But at least in Europe this is legally binding (every new installation has to obey these standards otherwise the electrician and/or the operator are liable for any damage or harm).
As far as I know these standards also apply to North Amerika.

chris1howell, 2020-10-01:
The standard in most of the world including USA is CCID 20. Which is AC
detection of ground fault at 20ma +0ma -5ma, so effectively the trip is
somewhere between 15ma and 20ma. There is no requirement for DC.

closed 2020-10-02 by faustomilletari

faustomilletari, 2020-10-02:
Hello,
thank you chris for your inputs. 

I still hear conflicting opinions about this. I have found a component (datasheet here https://vacuumschmelze.com/Assets-Web/4641-X900.pdf) which has the desired functionality and is specifically built for EVSE. 

Putting this component in the current design of OpenEVSE would mean to gut the GFCI portion of the circuit and simplifying it. Apparently PIN3 of the device linked above is low as long as there is no fault and goes in high-impedence whenever there is a system fault or a DC current >= 6ma or a AC current >= 30ma. 

We could remove everything and add a pull up resistor to the GFCI_INT line connected directly to the pin 3 of the GFCI fault device linked above.

For what concerns tests. the devices wants to sense pin number 1 low (GND) for any time between 30ms to 1.2 sec to trigger testing. I would imagine GFCI_TEST line is active high, so i was thinking to put a NOT gate in tail of that line (with a pull up resistor just to make sure), and drive the PIN1 of the GFCI device with that. 

The GFCI device absorbs around 35mA to function. Therefore i feel we can connect it to +5V and GND in the board. 

Do you think such a change would work. I am NOT knowledgeable in terms of electronics, i'm improvising. i'm awaiting you expert opinion!

Fausto

faustomilletari, 2020-10-02:
Something like this? (I'm really confident this WON'T work)

[image]

sauttefk, 2020-10-02:
@faustomilletari  [benvac 4641-X900](https://vacuumschmelze.com/Assets-Web/4641-X900.pdf) is a very interesting part, but not cheap (~80USD ). Seems to be a product developed by Vaccumschmelze (ferrite and magnets) and [Bender](https://www.bender.de/produkte/differenzstrom-ueberwachung/rcmb121) (electrical safety products). The product name benvac stongly suggests this.

But [WA RCM14-03](https://www.stegen.com/en/ev-products/126-residual-current-sensor.html) seems to be cheaper (~60 USD) and easier to connect.

faustomilletari, 2020-10-03:
In that case, with WA RCM14-03, we would directly connect the GFCI_INT (active high) to the fault pin and GFCI_TEST (active high) to the test pin. The power supply would be directly the +12 GND form the AC/DC unit.

Great!

sauttefk, 2020-10-05:
[KEMET FG-R05-3A](https://content.kemet.com/datasheets/KEM_SE0217_FG.pdf) is also a possible AC/DC ground leakage current sensor (~60USD).

jackpotdk, 2020-11-16:
Aha!! @faustomilletari , I am sitting with the exact same challenge here.
I am currently looking at https://www.bender.de/en/products/residual-current-monitoring/rcmb121

Have you made any progress on the matter?

DeltaVetal26, 2020-12-03:
Hi all,

There is a similar sensor from Stegen Electronics (Netherlands). 

https://www.stegen.com/en/ev-products/126-residual-current-sensor.html
It combines DC (6mA) and AC (30mA) leakage protection.

The only thing is that it requires an external 12V power supply and a logic level (for communicating with the controller) also 12V.
 
There are several options: 

1) There is an I2C bus on the OpenEVSE board, which is not used, but its pins are on the terminal. In the firmware, you can configure these pins for input-output and connect a sensor to them (if the one that I suggested is through a voltage divider (for example, an optocoupler). It is impossible without a divider, since the microcontroller is not designed for 12V from the sensor.

 Then just read the state of the sensor pin. If a leakage occurs, the sensor will change the state of the line, we will see it and will be able to stop charging.

2) If you have OpenEVSE together with ESP (WiFi), you can also connect the sensor to it (via a divider). In it, poll the status of the sensor and, in the event of a leak, send a RAPI command to OpenEVSE to stop charging. But here you need to make sure that the ESP always has a connection with OpenEVSE in order to be ready to stop charging.