OpenEVSE/OpenEVSE_PLUS #13: e-Golf Charging keeps starting/stopping

arnolde, 2020-10-09:
I have a 2020 VW e-Golf since April 2020 (new) and have been charging it since May with OpenEVSE with 2 phases (it has a 7.2kW onboard charger), it worked fine until about a week ago, then the car was only charged sometimes, with interruptions. Today I sat down and finally had time for an extensive diagnosis, and I see the following:
a) once in a while I see "diode check failed" on the OpenEVSE webUI page (without the cable having been touched - when I soft-reset the EVSE with the RAPI command, it starts working again)
b) when no error is displayed, the car charges for about 10 seconds and then pauses for 10-30sec (the only way I see that is by logging the power meter every second and watching the current in realtime) - there is no regular pattern though, and it seems not to change behaviour when I change the charge current between 6/10/16A in the webUI - see the many narrow, square spikes in the image (ignore the long steady part in the left half, that is not the car)

[image]

c) after about 5 tries the car seems to stop trying to charge

The car's smartphone app sometimes says "I'm sorry, there was an error of some sort, and charging stopped". If I click on "start charging" then it starts (a new sequence of atempts) again.

If I charge the car with VW's own 230V-charger (ICCB) it charges fine.

The only other info I have, is that a couple months ago the openEVSE type 2 plug got wet (the cap was on, it was dangling with the cap downwards, it got rained on, which collected inside the cap) and the CP pin got corroded. I cleaned it a little and charging worked again, reliably, until about 1 week ago. When I started diagnosing the charging problem this morning, I cleaned it thoroughly, but it did not improve this whole issue.

It almost seems to me like the car's firmware was updated a week ago (over the air, without my knowledge) and now it doesn't work well anymore with openEVSE? Is that possible?

arnolde, 2020-10-11:
@chris1howell Maybe I should have opened this under open_evse ? I'm sorry I'm not sure where exactly the right place for such an issue is.

Is there a way to read a debug log in realtime (like per UDP broadcast i.e.) so I can see if the openEvse is even detecting the interruptions at all? (I don't have current sensors installed) - then I could be reasonably sure that the car is causing it, unless the PWM is failing for some reason...

chris1howell, 2020-10-11:
Github issues are generally for bugs in coding or errors in design. Support
requests should be sent to support@openevse.com.

A. Diode check error while unplugged sounds like water in the handle making
connection from pilot to ground. This could also cause GFI errors.

B. This issue sounds like a poor CP or ground connection or power is not
detected by the vehicle when the relay closes. Ensure the relay is closing
properly and power is provided to the vehicle.

arnolde, 2020-10-12:
Thanks! The diode check does not fail when unplugged, only once in a while when trying to charge. I have GFCI disabled since I don't have a GFCI transformer mounted, I rely on my 30mA GFCI in my house's main distributor box.

I guess I will have to monitor the PWM and relay contacts and report back with those results.

Is there really no event log with timestamps I could access, so I can see if there were any rapidly changing events I'm missing on the ESP webUI?

arnolde, 2020-10-29:
Just to conclude with this problem: I opened up the type 2 plug handle and found water drops inside. Apparently the plug was not watertight :-( after drying out everything for a couple days indoors and then with a hairdryer to be sure, I reassembled everything and it now works again. My suspicion is that the e-Golf EV detected false currents somehow and interrupted the charging.