OpenEVSE/OpenEVSE_PLUS #14: Advanced_v5_2020.pdf missing power regulator

kallisti5, 2021-04-10:
@chris1howell   The new Advanced_v5_2020.pdf diagram is missing the power regulator. It shows 12v connected to 5v directly

chris1howell, 2021-04-10:
The power regulator is not missing. The new (February 2020) OpenEVSE ESP32
WiFi board has a built in 12v to 3.3v DC/DC converter. It is much more
efficient and simple to wire than the previous UBEC/Huzzah setup with
external 12v to 5v converter and onboard 5v to 3.3v regulator.

kallisti5, 2021-04-10:
oh.. Nice!  I just put the newer Feb 2020 board into my OpenEVSE and wired the wifi module up to the existing DC/DC converter.  Guess I need to crack it open and simplify a bit :-)

kallisti5, 2021-04-10:
although... the diagram does show 5v input pins on the wifi module which makes it a bit more confusing...
https://github.com/OpenEVSE/OpenEVSE_PLUS/blob/master/OpenEVSE_PLUS_v5/Advanced_v5_2020.pdf

SoDi1203, 2021-04-20:
hi, is this a broken link? 
https://github.com/OpenEVSE/OpenEVSE_PLUS/blob/master/OpenEVSE_PLUS_v5/Advanced_v5_2020.pdf