OpenEVSE/OpenEVSE_PLUS #18: add LICENSE

moedn, 2021-08-31:
Hey people! First of all: Great project! We're creating a meta search engine for open source hardware right now and scan a few well documented projects that have been recommended to us in order to promote them as such.

I found that a license file in this repo is missing. The README states that files are licensed under `CC BY-SA 3.0`. However, the license (and all other licenses I know so far) require that the legal code is distributed together with the licensed material. For that, the easiest may be to simply add [this plain text file](https://creativecommons.org/licenses/by-sa/3.0/legalcode.txt) as `LICENSE` to your repository.

However, you may (instead)

- update your license to CC-BY-SA-4.0 or
- shift to a hardware-specific license (like [CERN-OHL-S-2.0](https://ohwr.org/project/cernohl/blob/master/licence_texts/cern_ohl_s_v2.txt)) ← recommended

Let me know if I can help somewhere here :)

peace
-Moe

marcos-mendez, 2022-05-02:
@moedn Thanks for you suggestion i Really think this is also more like a Hardware license, but as this is not something that is related directly to a decission i can contribute i strongly suggest calling @chris1howell to that. Also a License is something that would protect OpenEVSE from bad usage from people or other companies. 

Maybe Chris we should discuss this in other place. Please, if you have any suggestion we are all open and at your order.

moedn, 2022-05-03:
sure, I'd be open for a call if I can be useful there :)

CERN OHL v2 is by far the best-maintained hardware-specific free/open license (from my perspective). Plus they offer the 3 top copyleft schemes.

If you have any (other) IP related questions or legal issues you're seeking advice for: At Open Source Ecology Germany e.V. (non-profit) [1] I'm working on a guideline for legal issues with 2 lawyers from Hamburg [2]. We just had a workshop on IPR last Friday [3], the recording will be uploaded soon.

If you want to join one of the coming workshops or want to ask some questions to our lawyers (or maybe even contribute to the guideline) → feel free to reach out to me :)

martin.haeuer [æt] ose-germany.de

peace

---

[1] \
[2] https://gitlab.com/OSEGermany/osh-guideline-legal-issues/-/issues\
[3] https://mobilize.berlin/events/d60059e5-6020-4f16-be61-2d7ab578a18c