OpenEVSE/OpenEVSE_PLUS #19: use releases instead of folders for versions

moedn, 2021-08-31:
Jo peeps :)

I suggest using git-tags to mark specific versions of the hardware instead of archiving copies of old versions in the root directory. (git saves _all_ versions anyway in the history)

That way each version of the hardware can be unambiguously referenced (like [OHLOOM](https://gitlab.com/OSEGermany/ohloom/-/tags) or [LibreSolar](https://github.com/LibreSolar/mppt-1210-hus/releases) (or FOSS projects in general) do).

GitHub offers some documentation for that [1], however, it's really just 3 clicks away.

Let me know if I can help you here with anything :)

marcos-mendez, 2022-05-02:
Hello @moedn I really understand you comment from a software overview, but on a Harware level architecture things are more Phisical and a complete understanding from previous version normally in mandatory for the entire system architecture and comprehenssion from the process. I Think this is also something that could be discussed, but is not a real Issue from the OPENEVSE_PLUS Repo and should be discussed somewhere else in the comunnity like the forum. https://openev.freshdesk.com/support/discussions/topics/new

Please recomend closing the issue here and opening a new discussion there. 

Thanks Marcos

moedn, 2022-05-03:
Thanks for the reply! This was just meant as a suggestion from a 'user perspective', in the end the process has to be smooth for you folks. If the current approach works well, there's no reason to change it :)

best\
-Moe

closed 2022-05-03 by moedn