OpenEVSE/OpenEVSE_PLUS #22: CCS-Type 2 DC from DC Power?

JohnGalt1717, 2022-01-05:
Hi, I'm building a DC Bus design that skips the entire inverter process. Is there any documentation on using DC fast charging with this project?

Basically what I'm looking at doing is using this, if possible to charge from a LiFePO4 battery + solar directly DC with a small DC/DC synchronous boost converter (at over 98%) which will cost about 1/8th the cost of a dc/ac converter and eliminate all of the losses from DC/AC and then back AC/DC in the car. The synchronous boost converter is easy (going for 384VDC) and I've built one. The harder stuff that is more risky is to talk to the car and now blow up a $30K battery in the car.

Thanks!

zymurgic, 2022-10-20:
CCS protocol for DC is very different from the simple PWM protocols that OpenEVSE does for AC charging to communicate with the vehicle's on-board charger. For a start, it will communicate via a powerline network over the CP pin.  There's a lot of reading to be done of the official standards.

I wish you all the best of luck, it's not going to be easy.

JohnGalt1717, 2022-10-20:
A couple of things here just if anyone else is thinking about this problem:

1. Tesla I believe uses CANBus which should be MUCH easier to work with.
2. ST has full sample units on their site that will do charging over CCS. Considering buying one because they come without the DC/DC converter.  The problem then becomes not the car, but communicating with the ST board to tell it max current draw etc.

strohrbaugh, 2023-02-24:
I'm interested in doing something similar, too. Have you developed this any further?

JohnGalt1717, 2023-02-24:
I've found the following:

Tesla was using CAN but now uses Power Communications and indeed uses the same protocol as CCS.
There are off the shelf Power Communications systems.

The basic flow would be:

1. Plug in
2. Negotiate (PP, PE etc.)
3. Establish communication over IP
4. Talk via IP to negotiate voltage and amperage.
5. End if PP/PE or IP request.

Charger would be:

1. DC in. (or PFC to DC)
2. DC/DC boost (or buck depending)
3. MCU that talks over IP and Power Communication Protocol that controls the boost converter.
4. Various sense pins for PP and PE.

The hardest part of this is actually the IP link and coms.

JohnGalt1717, 2023-02-25:
You’re still losing dc to ac and then ac to dc doing this and if you use an off the shelf transformerless inverter you’re not going to gain a thing. 

You’re at best at about 87% efficiency that way. 

The reason to do dc/dc is that you can get into the 97-98% efficiency with a synchronous buck/boost.

strohrbaugh, 2023-02-25:
Oh I completely agree, but I haven't yet solved the comms problem that you articulated above. The DC-AC solution "solves" it, but admittedly quite poorly. Let you know if I come across any improvements using something approximately off-the-shelf.

JohnGalt1717, 2023-02-25:
There's a company that the Electric Viking on YouTube referenced that has a DC/DC charging solution coming.  I can't find anything about them.  But it's a lead.

strohrbaugh, 2023-02-25:
Thanks, that's a good start.