OpenEVSE/OpenEVSE_PLUS #25: Time to ditch the 328P for an RP2040 or something purchasable?

DrFrankReade, 2022-11-02:
Leaving the certification problem for another time.

I feel it's time for a hardware refresh, esp. a new MCU, and maybe spread the board out a bit.

Can't buy the Arduinos anymore. I would not be surprised if the ATmega328 is extinct. Digikey only shows "marketplace" products which are in all likelihood counterfeit or pulls.  Plus, the existing code barely fits. 

I'm not kidding when I say we're using RP2040 at work in place of SAM chips we can't buy anymore, and always having a good laugh about it.  They require a few more external parts and they run at lower voltages, but they're impressive pieces of hardware with incredible community.  

Stock has got to be running low on the existing OpenEVSE boards to boot.

doanerock, 2022-12-31:
I am confused because the only microcontroller on the OpenEVSE is a esp32.  Are you referring to the OpenEVSE Programmer, because I am pretty sure you can use any FTDI programmer, so this is a non issue.

DrFrankReade, 2023-01-02:
[image]

Every OpenEVSE has one. 

It's where THIS code runs... 

[https://github.com/OpenEVSE/open_evse](url)

doanerock, 2023-01-03:
You are correct, my bad.  I had to bring up the schematic to find where it is located.  Looks like it is under the relay which is why I could not find it in the pictures.  
It looks like there are plenty of ATMEGA168P-20AU in stock.

DrFrankReade, 2023-01-03:
The 328 is the one that's shipping, and that's what the .bin releases are compiled for. The 168 also has half the program space, and if I recall correctly, the current code barely fits the 328. 

My OPINION, aside from easing the supply chain concerns, is that a heftier MCU would give a bit more headroom to put back the old LCD based UI, which I think is gone nowm, and missed by me at least, for redundancy's sake, because it's been my experience that when the ESP32 goes down, the charger goes squirrely and I either can or can't charge the car. 

Unfortunately, to move to a little ARM chip would require a complete hardware rework, as most stuff maxes out at 3.3V nowadays, and of course then supporting two hardware revisions... but if you can't buy chips, you can't ship product.  

There are other ATMEGA 8 bit chips - The 32U4 seems to have a lot of stock, and it's a popular chip, and there's also something called a atmega4808 and somebody's written a core for it too, although I have no idea how well it works. 

My thoughts anyway.

KanyonKris, 2024-04-14:
What a difference a year makes. While Microchip does not recommend the ATmega328P for new designs (https://www.microchip.com/en-us/product/atmega328p), it appears to be available for a reasonable price. Mouser has 90k in stock at $2.30 @ 100 qty.

The recommended successor, ATmega328PB is cheaper. It is not a drop-in replacement, there are 4 pin reassignments that will require a PCB revision, but 328 code will run on it. (https://www.pololu.com/file/0J1464/Atmel-42559-Differences-between-ATmega328P-and-ATmega328PB_ApplicationNote_AT15007.pdf)
Mouser has 33k of the extended temperature version in stock for $1.46 @ 100 quantity.

martiinezz, 2024-04-15:
Why not ESP32, it has 2 cores - WiFi and BT available - so it could be also possible to read cars battery state via BT to let say charge to 80 or 90% as some EVs lack such option.